eProsima/DDS-Router #120: DDS-Router and ROS2 nodes at the same device

mockbot, 2022-01-16:
How is it possible to start DDS-Router and ROS2 nodes at the same device ?
When trying to start DDS-Router and a node at the same device, I get some kind of shared memory access problem.
What is the preferred solution to avoid this issue ?

Scenario: ros2 robot in the remote lab and development/monitoring ros2 tools at homeoffice notebook

rsanchez15, 2022-01-17:
Hi @mockbot 

Could you give me more information about the error you are getting? It would also be good to know what is your network infrastructure, specifically if you are using any containerization technology such as Docker and what is the network configuration you are using (virtual network for the dockers, shared host network, etc.). 

Could you attach also the `.yml` configuration file that you are using to set up the DDS Router?

There is something else that is not entirely clear. On the one hand you say you are running everything on the same machine but you present a scenario with two LANs. I guess your use case is to connect both networks using the DDS Router right? Please correct me if I am wrong.

mockbot, 2022-01-17:
Hi,
I've simply started a node at the remote robot

***@***.****:*~*$ ros2 run demo_nodes_py listener

At another session at the same robot, I've started the ddsrouter
***@***.***:~/DDS-Router/install$ bin/ddsrouter -c robot-ddsrouter.yam
Starting DDS Router execution.
2022-01-17 14:44:28.096 [RTPS_TRANSPORT_SHM Error] Failed init_port
fastrtps_port7412: open_and_lock_file failed -> Function open_port_internal
2022-01-17 14:44:28.096 [RTPS_TRANSPORT_SHM Error] Failed init_port
fastrtps_port7413: open_and_lock_file failed -> Function open_port_internal
Starting DDS Router.

It is the same robot, no docker or any other stuff...

robot-ddsrouter.yaml

allowlist:

  - name: "/turtle1/cmd_vel"

  - name: "/turtle1/color_sensor"

  - name: "/turtle1/pose"

SimpleParticipant:

  type: local

  domain: 0

ServerWAN:

  type: wan

  id: 0

  listening-addresses:

    - ip: "79.201.239.133"

      port: 11711

      transport: "tcp"

LinuxWorkstationAtHome_with_ROS2_and_localDDSRouter---InternetRouterwithNAT=======Internet-WAN-Session=====>InternetRouter--LAN--WLAN--RoboterInTheRemoteLab_with_localDDSRouter

Question: is DDSRouter able to run at the same operating system instance or
does DDSRouter needs a separate LinuxBox ?

Thanks4All, maybe I have a total wrong idea of using ddsrouter

Cheers ;-)

mockbot, 2022-01-17:
or is it a ros2 foxy problem ?

closed 2022-01-17 by mockbot