eProsima/DDS-Router #190: ddsrouter-config.yaml

iekhanx, 2022-04-13:
The file " ddsrouter-config.yaml" cannot be located!! 
Can some provide me this file , please.
thank you,

jparisu, 2022-04-13:
Hi @iekhanx 
This is a file that must be created by the user for its specific use of the DDS-Router.
Check this documentation in order to learn how to write it: https://eprosima-dds-router.readthedocs.io/en/latest/rst/user_manual/configuration.html

There are some examples of configuration files that installed along the project that may be of help in order to create you own: https://github.com/eProsima/DDS-Router/tree/main/resources/configurations/examples

rsanchez15, 2022-04-20:
Hi @iekhanx 

Is this issue related to issue #193? If so, since the problem has been solved, please let us know or close this issue. 

Thank you very much.

rsanchez15, 2025-02-14:
Since we haven’t heard back from you, I'm closing this issue due to inactivity. If you're still experiencing this problem or have additional information to share, please leave a comment or reopen the issue.

closed 2025-02-14 by rsanchez15