eProsima/DDS-Router #191: Procedure replication for global installation of DDS-Router

iekhanx, 2022-04-13:
Hi 
I am trying to replicate the procedure guided by following link:
https://eprosima-dds-router.readthedocs.io/en/latest/rst/developer_manual/installation/sources/linux.html#developer-manual-installation-sources-linux

But when I execute the following section:

**3. Once all dependencies are installed, install DDS Router:
cd ~/DDS-Router
git clone https://github.com/eProsima/DDS-Router.git
mkdir DDS-Router/build
cd DDS-Router/build
cmake .. -DCMAKE_INSTALL_PREFIX=~/DDS-Router/install -DCMAKE_PREFIX_PATH=~/DDS-Router/install
cmake --build . --target install**
The resulted in following error:

**$ cmake .. -DCMAKE_INSTALL_PREFIX=~/DDS-Router/install -DCMAKE_PREFIX_PATH=~/DDS-Router/install
CMake Error: The source directory "/home/imrank/DDS-Router/DDS-Router" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.**

Can you please resolve the issue  although cloned from provided link but CMaketext.txt file is missing.
Please treat as an urgent.
Thank you

iekhanx, 2022-04-13:
I noticed that the procedure below is using CMAKE or cmake which only applies to following three modules
foonathan_memory_vendor
Fast-DDS
Fast-CDS
https://eprosima-dds-router.readthedocs.io/en/latest/rst/developer_manual/installation/sources/linux.html#developer-manual-installation-sources-linux

For DDS-Router require Python scripted procedure to build as follow:
https://github.com/eProsima/DDS-Router

please make clear in your README.md file. 
There is still issue of binding as an integral part of the all FOUR 
DDS-Router
Fast-DDS
Fast-CDS
foonathan_memory_vendor

remain unexplained

iekhanx, 2022-04-13:
jparisu
Can you give me your first hands experience that DDS-Router is running on the Server?
After issuing commands within its own directory i.e. DDS-Router/install/ddsrouter_tool/bin/ddsrouter DDS_ROUTER_CONFIGURATION.yaml

./ddsrouter DDS_ROUTER_CONFIGURATION.yaml 
Starting DDS Router Tool execution.
Usage: Fast DDS Router 
Connect different DDS networks via DDS through LAN or WAN.
It will build a communication bridge between the different Participants included in the provided configuration file.
To stop the execution gracefully use SIGINT (C^) or SIGTERM (kill) signals.
General options:
  -h --help         Print this help message.
  -c --config-path  Path to the Configuration File (yaml format) [Default: ./DDS_ROUTER_CONFIGURATION.yaml].
  -r --reload-time  Time period in seconds to reload configuration file. This is needed when FileWatcher functionality is not available (e.g. config file is a symbolic link). Value
                    0 does not reload file. [Default: 0].
  -d --debug        Activate debug Logs (be aware that some logs may require specific CMAKE compilation options).
2022-04-13 18:22:07.377 [DDSROUTER_ARGS Error] ERROR: Unknown argument: <DDS_ROUTER_CONFIGURATION.yaml>. -> Function parse_arguments
:~/DDS-Router/install/ddsrouter_tool/bin$ l
total 136
drwxr-xr-x 2 imrank imrank   4096 Apr 13 18:21 .
drwxr-xr-x 4 imrank imrank   4096 Apr 13 18:02 ..
-rwxr-xr-x 1 imrank imrank 125512 Apr 13 18:02 ddsrouter
-rw-r--r-- 1 imrank imrank   3551 Apr 13 18:19 DDS_ROUTER_CONFIGURATION.yaml
~/DDS-Router/install/ddsrouter_tool/bin$ ./ddsrouter 
Starting DDS Router Tool execution.
Not configuration file given, using default file DDS_ROUTER_CONFIGURATION.yaml.
DDS Router running.

I randomly picked a file from resource/configuration/examples/wan_server.yaml renamed the file as
DDS_ROUTER_CONFIGURATION.yaml and submitted to ddsrouter at command line
but it failed to read from the file, I wonder why?
then simply typed ddsrouter and enter and its say ddsrouter running.  
What next to do to get going ?

jparisu, 2022-04-19:
Hi @iekhanx 

I assume you have already managed to avoid these issues mentioned here for the comments you do in this issue: https://github.com/eProsima/DDS-Router/issues/193

If you still have some troubles with this, I strongly suggest to check our public and online documentation. Here it is a getting started user friendly tutorial: https://eprosima-dds-router.readthedocs.io/en/latest/rst/getting_started/usage_example.html

closed 2025-02-14 by rsanchez15