eProsima/DDS-Router #193: DDS-Router issue

iekhanx, 2022-04-14:
Hi there,
I managed to build DDS-Router and is up and running using DDS_ROUTER_CONFIGURATION.yaml for Server
and DDS_ROUTER_CONFIGURATION.yaml for Client as in attached Pngs file the files take take from ../../example 
wan_server.yaml and wan_client.yaml 
I try to send across the <HelloWorldTopic> ; <HelloWorld>  it seem to be NOT working
can you please advise accordingly. to get working?

[image]
[image]

jparisu, 2022-04-19:
Hi @iekhanx 

I am glad you managed to run the DDS Router. Please, close those issues or discussions that do not longer require support.

I am sorry but I do not understand your screenshots. The DDS Router is a `passive` application, it does not interact with the user. This means, once running it will route every DDS message recieved from one participant to the other participants.
I think you are trying to write interact with the application by terminal commands, but this is not supported; the configuration of the topics used comes from the Configuration file you are using.

I strongly suggest you to check our documentation, with tutorials, examples and complete details on the execution of the DDS Router: https://eprosima-dds-router.readthedocs.io/en/latest/rst/getting_started/usage_example.html

jparisu, 2022-04-19:
Hi @iekhanx 

Regarding your multiple issues, I would like to redirect you to our support team: support@eProsima.com
They will be glad to hear about your use case, your specific issues, and you could scheduled targeted support for your project.

Have in mind that we develop open source code, and we are happy to help our community, but we do not offer free support for specific cases further than answering questions and correcting bugs.

iekhanx, 2022-04-19:
I am sending these messages across Three PCs 
1/ Publisher
2/ Subscriber   - Ubuntu 22.04v
3/ Window-10

rsanchez15, 2022-04-20:
Hi @iekhanx 

I am closing this issue as the problem has been solved. 

As this was not a DDS Router issue, if you have any other similar issues/questions/concerns in the future, please visit the [Discussions](https://github.com/eProsima/DDS-Router/discussions) page of this repository and open a new discussion there in the category you think best describes your problem.

closed 2022-04-20 by rsanchez15

iekhanx, 2022-04-20:
Although my problem is some how with the help of you lot managed to solved but there are plenty of disarray and update in documents need to be adjusted and corrected as I do not have rights to edit but there are mis match in the documents what document Say on github and what says on eProsima.com web site a coherent consistency must exists besides details elaborate replication procedure must exist and distinction must be given Linux separately and Windows Separately current document collide each other if user following Linux procedure then it MUST end and go NO further! 
If user following Windows procedure it must end and GO no further.
1/ Procedure for GitHub down load and build must be separated from eProsima.com Binary version down loads.
2/ if User down load from eProsima.com (binary version install.sh) then this procedure must have begin n stop after install
3/ DDS-Router procedure must have from source install hence it must exists on GitHub only so installer may know he has to prepare system from the scratch to configure then download DDS-Router then build the router distinctly for chosen OS.
4/ There are many anomaly I discovered in the documentation I can't recall all of them right now but main advise is:
5/ to lets say Binary Install for Fast-DDS; one procedure ON Linux open and guide the user to its finish.
6/secondly Fast-DDS from source build complete procedure for Linux start and finish.
7/ Third Fast-DDS for MS Window start and finish
 8/ Similar path should adopted to rest of other modules like DDS-Router
9/ Code generator must be separate procedure n identify as a Tool and it usage although if user stumbled on youtube.com 
then he may discover the usage of these tools hence, link at the procedure must be placed.
I guess here we try to make life easy n untangled web instead of making complicated? These are few tips.
Thank you,
IK

**Remember:** 
There is a file IS missing in Fast-DDS project examples\ optionparser.hpp when users try to build ALL given examples  the build fails.
An example for DDS-Router's are placed in Fast-DDS project Library?  IF DDS-Router download from the source then DDS-Router example should be go along with the project too I guess?