eProsima/DDS-Router #195: DDS-Router for Windows 10 build failed

iekhanx, 2022-04-17:
Hi there,
I am encountering a single issue within DDS-Router build for WINDOW-10 OS version.
 The compiling process succeed but one module failed to get build within multiple
modules, that "ddsrouter-yaml-validator" sems to failed as following log depict:

Can anyone give a clue what is missing from the build?

Your reply is greatly appreciated.
Thank you,
IK
==============Error log=================
C:\Users\imrank\DDS-Router>colcon build
Starting >>> ddsrouter_cmake
Starting >>> YAML_CPP
Starting >>> ddsrouter_yaml_validator
--- stderr: ddsrouter_yaml_validator
c:\users\imrank\appdata\local\programs\python\python36-32\lib\site-packages\setuptools\command\install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< ddsrouter_yaml_validator [8.03s]
Finished <<< ddsrouter_cmake [10.0s]
Starting >>> ddsrouter_utils
Starting >>> ddsrouter_docs
Finished <<< ddsrouter_utils [8.48s]
Starting >>> ddsrouter_event
--- stderr: YAML_CPP
CMake Deprecation Warning at test/gtest-1.10.0/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
CMake Deprecation Warning at test/gtest-1.10.0/googlemock/CMakeLists.txt:45 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
CMake Deprecation Warning at test/gtest-1.10.0/googletest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
---
Finished <<< YAML_CPP [29.3s]
Finished <<< ddsrouter_event [12.0s]
Starting >>> ddsrouter_core
Finished <<< ddsrouter_core [4.38s]
Starting >>> ddsrouter_yaml
--- stderr: ddsrouter_yaml
CMake Error at C:/Users/imrank/DDS-Router/install/ddsrouter_cmake/share/ddsrouter_cmake/cmake/cpp_common/find_external.cmake:26 (find_package):
  By not providing "Findyaml-cpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "yaml-cpp",
  but CMake did not find one.
  Could not find a package configuration file provided by "yaml-cpp" with any
  of the following names:

    yaml-cppConfig.cmake
    yaml-cpp-config.cmake

  Add the installation prefix of "yaml-cpp" to CMAKE_PREFIX_PATH or set
  "yaml-cpp_DIR" to a directory containing one of the above files.  If
  "yaml-cpp" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  C:/Users/imrank/DDS-Router/install/ddsrouter_cmake/share/ddsrouter_cmake/cmake/cpp_common/configure_project_cpp.cmake:51 (find_external_projects)
  CMakeLists.txt:56 (configure_project_cpp)
---
Failed   <<< ddsrouter_yaml [1.03s, exited with code 1]
Aborted  <<< ddsrouter_docs [26.5s]

Summary: 6 packages finished [39.7s]
  1 package failed: ddsrouter_yaml
  1 package aborted: ddsrouter_docs
  3 packages had stderr output: YAML_CPP ddsrouter_yaml ddsrouter_yaml_validator
  1 package not processed

jparisu, 2022-04-19:
Hi @iekhanx 

Some points:

- It seems like you are using an old version of CMake. I strongly suggest to update to a 3.x version
- The error you are encountering is that yaml-cpp is not well installed or it is not found from colcon to build the forward packages.
  - Check our documentation on how to install yaml-cpp for windows using CMake: https://eprosima-dds-router.readthedocs.io/en/latest/rst/developer_manual/installation/sources/windows.html#yaml-cpp
  - use `colcon` argument `--event-handlers console_direct+` in order to see the logs of the compilation. With that, you could check if the library is being installed correctly and if the installed files are in the PATH to compile the `ddsrouter_yaml`.

Colcon it is not so stable for Windows, my suggestion is to install `yaml-cpp` in an aside workspace (using CMake for example) and then adding it to the path when compiling the rest of the library

rsanchez15, 2025-02-14:
Since we haven’t heard back from you, I'm closing this issue due to inactivity. If you're still experiencing this problem or have additional information to share, please leave a comment or reopen the issue.

closed 2025-02-14 by rsanchez15