eProsima/DDS-Router #256: Configuration example: WAN with dynamic ipaddresses  and dyndns (optional SSL/TLS)

mockbot, 2022-07-29:
Hi,
is there a real world internet wan configuration for home labs available ?

1. both local networks (192.168.1.0/24 and 192.168.2.0/24) are connected to the internet via provider router, using NAT and dynamic router ip addresses
2. both endpoint are reachable  by using dyndns FQDN (dds1.dyndns.org and dds2.dyndns.org)

rviz2-manager<--->dds-gw1--->router1(dds1.dyndns.org)<---internet--->router2(dds2.dyndns.org)<---dds-gw2<--->rover

How is this possible ?
What are the keywords for using this scenario ? 
Is SSL/TLS secure communication between both dds-routers available (self signed certificates) ? 

I guess, such wan-home-example.yaml is very useful for all homeworkers and students.

jparisu, 2022-08-12:
Hi @mockbot 

I am not sure what do you mean about "real world internet wan configuration for home labs", could you be more precise?
In the documentation, this example is a "real world internet wan configuration" and **explains how to set-up  an standard internet router** so the communication could be established from home by **port-forwarding**: https://eprosima-dds-router.readthedocs.io/en/latest/rst/user_manual/wan_configuration.html
This configuration could be found here in the repository: https://github.com/eProsima/DDS-Router/blob/0.4.0/resources/configurations/examples/wan_client.yaml

**Attention** be careful if using `main` branch, as we are updating the configuration (it will add a new configuration version `v3.0` so old executions work as they used to.

We have also tested dyndns and it works properly. It is possible to set a "Domain Name (DNS)" rather than an IP easily in the configuration. Check this documentation: https://eprosima-dds-router.readthedocs.io/en/latest/rst/user_manual/configuration.html#network-address

mockbot, 2022-08-13:
ReHi,
"real world internet wan configuration for home labs"...
Oh yes, I'm a networking guy and I have problems to understand the usage of your keywords...

e.g.
ip means internet protocol, but you should use ipaddress. Next issue, you use the domain keywords for an identifier of a domain and as ipaddress when using a FQDN and TCP layer 3 protocol instead of a dotted ip address number...
Why do you have 2 different keywords for the partner ipaddress ? Confusing...
ipaddress should handle dotted ipaddress numbers and FQDN. The standard resolver libs getHostByName will handle it since the beginning ;-)

What is "real world internet wan configuration for home labs" ?
1. both ends are using dyndns and NAT, no hardcoded ipaddresses  numbers.
2. communication protocol between both partners TCP/SSL

I've tried it with version from June 2022, but I've failed.  Topics were not visible and I was not able to see open sockets. I guess, I need to go deeper into the traffic with wireshark tools.

domain is OK in the DDS contect, but

mockbot, 2022-08-15:
I've prepared a small documentation here: https://ns.xdsnet.de/extCloud/nextcloud/index.php/s/qYJWBoZEpAPKCyG

jparisu, 2022-08-17:
Hi @mockbot 
Thank you for the detail explanation.

## IP and Domain
We separated `ip` from `domain` tag because we thought it would be easier to understand for a user. It seems is not though, so we may change it in future releases.
We appreciate a lot the users feedback so we can improve the user experience of the DDS Router.

## Real World Scenario
This scenario is actually a pretty good example on how to use the new feature `Repeater`.
This is a feature that we are working on, and will be released in next version (it is already in `main`), and it allows to deploy a `TURN` server to communicate multiple points of the internet within each other (including the own server, that could communicate its own network: *LAB*).
Here it is the documentation available for the Repeater so far: https://eprosima-dds-router.readthedocs.io/en/latest/rst/use_cases/repeater.html
Handling the FQDN, ports and TCP is as easy as using a configuration like this one:
https://eprosima-dds-router.readthedocs.io/en/latest/rst/examples/repeater_example.html#client-participants