eProsima/DDS-Router #275: colcon build failed

mockbot, 2022-08-16:
during installation at a fresh environment, I've noticed, that 'colcon build' failed with some errors...
assert verb_extensions, 'No verb extensions'

after some search I found this solution:

pip3 install pytest-rerunfailures

cd ~/DDS-Router
rm -rf src

mkdir src
vcs import src < ddsrouter.repos
colcon build

Bug or feature ? ;-)

jparisu, 2022-08-17:
Thank you for reporting this issue.
It is true it seems like a common known error when using colcon:
https://github.com/aws-robotics/aws-iot-bridge-example/issues/2

This is most likely a bug (colcon bug).
We will add for future versions a `requirements.txt` so it can be executed in a virtual environment, or add this dependency to the installation guide.

However, python is only used in `ddsrouter_yaml_validator` and in some tests.
So you can use as well `colcon build --packages-skip ddsrouter_yaml_validator` to build it without problem.

rsanchez15, 2025-02-14:
Hi @mockbot 

Since we haven’t heard back from you, I'm closing this issue due to inactivity. If you're still experiencing this problem or have additional information to share, please leave a comment, and I'll be happy to reopen the issue.

closed 2025-02-14 by rsanchez15