eProsima/DDS-Router #276: Messages not routing with Foxy

lairdm, 2022-08-16:
I've spent about a week debugging this and have come to the conclusion that messages sent from Foxy aren't getting routed, I'm not sure if that's in general or just over TCP.

I've setup the repeater scenario from the documentation with two router instances on separate internet connections and a cloud router out in a GCP VM. I'm sending a simple twist message from one side.

`ros2` topic pub  /cmd_vel geometry_msgs/msg/Twist '{linear: {x: -0.5, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1}}'

If the Twist is sent from a laptop or VM running Foxy it's not forwarded through the local DDS Router to the cloud instance. If it's sent via a Humble instance the message forwards. I've set all instances up with a local echo participant and can see the writers across the three instances for Humble, but only the local one when sending from Foxy. I've also used RTI's Admin Console and I can see in the local instance that with both Foxy and Humble the DDS Router subscribes to the topic, it just doesn't forward messages.

I suspect it's a QoS issue, however I can see from the echo participant that both Foxy and Humble are sending with Transient Local and Reliable. I'm somewhat new to ROS and DDS, but I've seen other hints in forums there was some kind of change in DDS and the default QoS between Foxy and Galactic. I'm hoping this is a known issue with an easy fix. Thanks.

jparisu, 2022-08-17:
Hi @lairdm 

Glad to know you are using our new `Repeater` feature. Every feedback is welcome.

Regarding your issue, I was not able to replicate the error, nor in foxy neither humble. I have checked the QoS and they have not changed for the command you are using. And this should not be a problem, as the DDS Router currently forward all the data (the internal entities use the lowest restriction QoS).

We have recently added a new functionality to `EchoParticipant` where you can see the Discovery information (and set a `verbose` option). From this PR https://github.com/eProsima/DDS-Router/pull/264 it is available and it may help you to see if the problem is that messages are not being forwarded correctly, or it is a discovery problem.
Sorry, but this feature is not well documented yet. I am working on it and I would upload it ASAP.

So, in order to help you with your issue, I may need to know if the error occurs in discovery (check that Cloud and Edge2 DDS Routers create the entities regarding the Edge1, and Discover the remote endpoitns) or whether it is a communication problem (EchoParticipant may not show data been forwarded from a router to the other).

lairdm, 2022-08-17:
I've solved the issue but I have no idea why it works. I briefly got a Foxy VM working, which confused me. I then rebooted it and it stopped working. Digging through the `data: true` and `verbose: true` settings for the echo participant, thank you that was very useful, I saw that DDS Router wasn't even seeing the twist commands being sent locally on Foxy.

I found this post talking about disabling shared memory [1]. I tried adding the profile and now the Foxy node works perfectly fine. And again, this is something I've sort of gotten hints about as I've been learning and evaluating RTI's Routing Service over the last month. I've seen hints that something with QoS or the SHMEM magic number/identifier changing between DDS versions around the time of Foxy/Galactic. I don't fully understand but I've seen different mentions about Foxy and SHMEM compatibility and changes.

Regardless, I've found a way to make Foxy work consistently. Thank you very much for your assistance.

[1] https://answers.ros.org/question/370595/ros2-foxy-nodes-cant-communicate-through-docker-container-border/

jparisu, 2022-08-18:
Hi @lairdm 
I am glad that it works now. We are working to improve our documentation, but indeed the Echo feature is a great advantage.

Regarding the changes in ROS2 that you mention, I am not completely aware of it, but notice this;

- Foxy works (by default) with eProsima Fast DDS. Galactic has Cyclone DDS by default. So this is a great change in middleware, especially regarding default values. But the default QoS should have no change as far as I know.
- Shared Memory Transport is not something standard in DDS, so each implementation has its own (if they have one). In Fast DDS we know that it has been issues when working with dockers, as entities think they are in the same host while they do not have access to the same shared memory segment. This is configurable via CMake options, via code or by xml configuration.
- DDS Router always works with Fast DDS underneath. Thus, it will behave differently with Fast DDS entities than with other DDS vendors entities. However, this differences would be noticed in performance, but should not affect the user experience (except this SHM docker failure).

Please, if this issue has been solved, close this issue and feel free to open it or open a new one if any other problem arises.

closed 2022-08-18 by lairdm