eProsima/DDS-Router #298: docs: `addresses` doesn't seem valid anymore in `connection-addresses`

russkel, 2022-09-12:
I am running ddsrouter on the main trunk. I based my configuration off the latest docs:

https://github.com/eProsima/DDS-Router/blame/main/docs/rst/user_manual/wan_configuration.rst#L161

Running the config I get:
```
 Error loading DDS Router configuration from yaml:
 Error reading yaml list under tag <connection-addresses> :
 Error reading yaml sequence of type <N8eprosima9ddsrouter4core5types7AddressE> :
 Address requires to specify <ip> or <domain>. -> Function main
```

Only after I removed the `addresses` key and restructured it like so:

```yaml
      connection-addresses:
       # - discovery-server-guid:
       #     id: 2                         # Id of the Discovery Server of A
          - ip: 1.1.1.1                 # Public IP of Ha
            port: 11666                 # Port forwarded in Ra
```

Was I able to get it to work.

jparisu, 2022-09-12:
Hi @russkel ,

You are completely right.
This is due to the **YAML Configuration** version in use:

## Reason

We are constantly adding new features to the DDS Router, and thus the configuration files change. This is why there is a tag `version` that allows your configuration to be correctly parsed and do not break backwards compatibility when new features are added.

You are probably using the latest version `v3` that does not need a tag `addresses` under `connection-addresses`, while the **documentation is still in an older version** where `v2` is explained and `connection-addresses` requires an `address` tag.

## How to solve
Just add the tag `version: v2` to the base of your yaml configuration so you are correctly using the configuration version that you want.
Or update your configuration to the `v3` version (not suggested as it is still not correctly documented. However this is the only main change added).

## Conclusion
This is not a bug, it is just a mismatch between the code and the documentation versions.
We are sorry for the inconvenient and we are always happy to help.
If you want a more detailed answer regarding your use case, please share your `.yaml` configuration file and we can advice you on how to design it. 

---

## Why these changes
If you are curious about why this tag is not needed anymore, it is because we now consider a `wan` participant as _Initial Peers_, not as _Discovery Server_, so it does not require so much configuration.
However, using tag `kind: wan_ds` will still use `DiscoveryServer` and the old configuration will work.

## Versions
Be careful to be coherent when checking the documentation and use the application.
We suggest to use last release version, and not `main` branch, because some features could be not completely documented in `main` documentation.

### Last release
v0.4.0
Application: https://github.com/eProsima/DDS-Router/tree/0.4.0  branch: `0.4.0`
Documentation: https://eprosima-dds-router.readthedocs.io/en/v0.4.0/

russkel, 2022-09-12:
> This is not a bug, it is just a mismatch between the code and the documentation versions.
> We are sorry for the inconvenient and we are always happy to help.

Thanks @jparisu for the comprehensive reply. I understand it's a work in progress and no inconvenience was caused. I only wanted to report what may have been an oversight.

closed 2022-09-12 by russkel