eProsima/DDS-Router #349: run example error

mengchaoheng, 2023-02-05:
topic can't listening by `ros2 run demo_nodes_cpp listener`

```Console
parallels@ubuntu-linux-20-04-desktop:~/DDS-Router/install/ddsrouter_tool/share/resources/configurations/examples$ ddsrouter -c change_domain_allowlist.yaml 
Starting DDS Router Tool execution.
DDS Router running.

```
```Console
parallels@ubuntu-linux-20-04-desktop:~/FastDDSGenHelloWorld/build$ ROS_DOMAIN_ID=0 ros2 run demo_nodes_cpp talker
[INFO] [1675612035.867491964] [talker]: Publishing: 'Hello World: 1'
[INFO] [1675612036.867868934] [talker]: Publishing: 'Hello World: 2'
[INFO] [1675612037.870178882] [talker]: Publishing: 'Hello World: 3'
[INFO] [1675612038.870980998] [talker]: Publishing: 'Hello World: 4'
[INFO] [1675612039.869903435] [talker]: Publishing: 'Hello World: 5'
[INFO] [1675612040.870213935] [talker]: Publishing: 'Hello World: 6'
[INFO] [1675612041.874682112] [talker]: Publishing: 'Hello World: 7'
[INFO] [1675612042.875003580] [talker]: Publishing: 'Hello World: 8'
[INFO] [1675612043.875383813] [talker]: Publishing: 'Hello World: 9'
```

```Console
parallels@ubuntu-linux-20-04-desktop:~/FastDDSGenHelloWorld/build$ ROS_DOMAIN_ID=1 ros2 run demo_nodes_cpp listener
2023-02-05 23:47:18.643 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7662: open_and_lock_file failed -> Function open_port_internal
2023-02-05 23:47:18.643 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7663: open_and_lock_file failed -> Function open_port_internal

```

mengchaoheng, 2023-02-05:
the router file is :
```Console
########################################
# CHANGE DOMAIN WITH ALLOWLIST EXAMPLE #
########################################

##################################
# CONFIGURATION VERSION
version: v3.0                                                               # 0

##################################
# ALLOWED TOPICS
# Allowing FastDDS and ROS2 HelloWorld demo examples topics

allowlist:
  - name: HelloWorldTopic                                                   # 1
    type: HelloWorld                                                        # 1
  - name: rt/chatter                                                        # 2
    type: std_msgs::msg::dds_::String_

##################################
# PARTICIPANTS
participants:

##################################
# SIMPLE PARTICIPANT DOMAIN 0
# This participant subscribes to allowlist topics in DDS Domain 0 and listen every message published in such DDS Domain

  - name: SimpleParticipant_domain0                                         # 3
    kind: local                                                             # 4
    domain: 0                                                               # 5

##################################
# SIMPLE PARTICIPANT DOMAIN 1
# This participant subscribes to allowlist topics in DDS Domain 1 and listen every message published in such DDS Domain

  - name: SimpleParticipant_domain1                                         # 6
    kind: local                                                             # 7
    domain: 1                                                               # 8

##################################
# CONFIGURATION DESCRIPTION

# This configuration example configures a DDS Router to listen to every message published in two different domains
# and transmit those messages through the other domain.

# 0: Use YAML configuration version v3.0

# 1: Allow DDS Topic Name <HelloWorldTopic> with type <HelloWorld>.

# 2: Insert new topics in order to route to them.

# 3: New Participant with name <SimpleParticipant_domain0>.

# 4: Kind of SimpleParticipant_domain0: <local>.
# LAN UDP communication with default simple multicast discovery.

# 5: SimpleParticipant_domain0 will use DDS Domain ID <0>.

# 6: New Participant with name <SimpleParticipant_domain1>.

# 7: Kind of SimpleParticipant_domain1: <local>.

# 8: SimpleParticipant_domain1 will use DDS Domain ID <1>.
```

mengchaoheng, 2023-02-06:
**which version of DDS-Router are you using?** 
```Console
parallels@ubuntu-linux-20-04-desktop:~/DDS-Router/install/ddsrouter_tool/share/resources/configurations/examples$ ddsrouter -v
DDSRouter v1.1.0
commit hash: 86e2241347887ef3438a0ec35611feae10b3a80f
```
**What about your ROS 2 setup?**

the ros2 version is foxy on

```Console
resources/configurations/examples$ uname --all
Linux ubuntu-linux-20-04-desktop 5.15.0-58-generic #64~20.04.1-Ubuntu SMP Fri Jan 6 16:45:34 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
```
and fast dds is install 2.9.1 version and fastddsgen is
```Console
parallels@ubuntu-linux-20-04-desktop:~/DDS-Router/install/ddsrouter_tool/share/resources/configurations/examples$ fastddsgen -v
openjdk version "13.0.7" 2021-04-20
OpenJDK Runtime Environment (build 13.0.7+5-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 13.0.7+5-Ubuntu-0ubuntu120.04, mixed mode)
```
and, the .bashrc file is 
```Console
# dds
source ~/Fast-DDS/install/setup.bash # have to , this is Colcon installation
export LD_LIBRARY_PATH=~/test_build/Fast-DDS/install/lib/ #why Could not find a package configuration file provided by "fastcdr"? so I command this line. this is CMake installation.
# gen
export PATH=/home/parallels/Fast-DDS/src/fastddsgen/scripts/:$PATH
export PATH=/home/parallels/Micro-XRCE-DDS-Gen/scripts/:$PATH
# dds-router
source ~/DDS-Router/install/setup.bash
export PATH=/home/parallels/DDS-Router/install/ddsrouter_tool/bin:$PATH
# some install lib
export PATH=/usr/local/lib/:$PATH

# ros2
source /opt/ros/foxy/setup.bash
```

and I really don't know how to deal with the relationship between the various versions of dds. For example, I installed fast dds when I installed ros2, and then I installed it, and then I installed the latest version 2.9.1, and source ~/Fast-DDS/install/setup .bash , finally I found an error when running the case, so I had to export LD_LIBRARY_PATH=~/test_build/Fast-DDS/install/lib/

juanlofer-eprosima, 2023-02-06:
Thank you for the report. Indeed, it seems that you have compiled DDS-Router `main` branch. If possible, please test my suggestion (build DDS-Router `v1.1.0` and Fast-DDS `v2.9.0`) before digging any further.

Cheers

rsanchez15, 2025-02-14:
Hi @mengchaoheng 

Since we haven’t heard back from you, I'm closing this issue due to inactivity. If you're still experiencing this problem or have additional information to share, please leave a comment, and I'll be happy to reopen the issue.

closed 2025-02-14 by rsanchez15