eProsima/DDS-Router #380: Adding ROS 2 introspection as SUPER_CLIENT

royito55, 2023-06-06:
Hi,

I have managed to connect two robots running a discovery server each by running a DDS router from a 3rd PC. Here is the config YAML:
```
version: v3.0

participants:

  - name: ROS_2_Domain_0
    kind: simple
    domain: 0

  - name: ROS_2_DS_Client_1
    kind: discovery-server

    discovery-server-guid:
      ros-discovery-server: true
      id: 1

    connection-addresses:
      - discovery-server-guid:
          ros-discovery-server: true
          id: 0
        addresses:
          - domain: 192.168.1.72
            port: 11811

  - name: ROS_2_DS_Client_2
    kind: discovery-server

    discovery-server-guid:
      ros-discovery-server: true
      id: 2

    connection-addresses:
      - discovery-server-guid:
          ros-discovery-server: true
          id: 0
        addresses:
          - domain: 192.168.1.190
            port: 11811
```

However, I also want to be able to see all of the topics offered by both robots from the 3rd PC, so is there like a super client configuration I can add to it so it can discover all of them? I tried with the default super client XML, but I realize there is no discovery server (DDS router instead) running in this 3rd PC so it makes sense that it doesn't work:
```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <builtin>
                    <discovery_config>
                        <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                        <discoveryServersList>
                            <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
                                <metatrafficUnicastLocatorList>
                                    <locator>
                                        <udpv4>
                                            <address>127.0.0.1</address>
                                            <port>11811</port>
                                        </udpv4>
                                    </locator>
                                </metatrafficUnicastLocatorList>
                            </RemoteServer>
                        </discoveryServersList>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>
    </profiles>
</dds>
```

I noticed that if I echo a topic in either of the robots, it automatically appears on the 3rd PC. Which is great, but I'd like to see them all from the first time I list the topics.

Here's a diagram to illustrate:

[image]

jparisu, 2023-06-06:
Hi @royito55 

Your configuration file seems fine to your scenario.
What you are experiencing here is a performance feature in DDS Router that makes the router to **no create endpoints until there is a DataReader (ROS2 subscription)** in the network. This reduces the amount of DDS Entities that are created in the DDS Router application and the traffic it routes, and thus improves the efficiency.

Nevertheless, we have lately seen the necessity to add a configuration feature that allows to disable such feature.
Once this feature is implemented and uploaded we will notify you.

## Extra comment

There is no need to use XML configuration in your `fastdds discovery` application, as it is already a Server, so setting it as SuperClient will have no effect.

rsanchez15, 2025-02-14:
Hi @royito55 

Since we haven’t heard back from you, I'm closing this issue due to inactivity. If you're still experiencing this problem or have additional information to share, please leave a comment, and I'll be happy to reopen the issue.

closed 2025-02-14 by rsanchez15