eProsima/DDS-Router #412: DDS Router not forwarding ros2 messages before the first echo.

Woojin-Crive, 2023-12-04:
I want to use DDS router to make a wan communication.
I have a public ip port-forwarded pc(server). and a pc behind a nat(client).
They are not in the same network.
dds router container is running at the server and client each.

But when i exec `ros2 run demo_nodes_cpp talker` in the client container, below message is printed out at the console and no more message is printed.
After that, If i exec `ros2 run demo_nodes_cpp listener` in the same container, every message is printed at the console. And i can check it at the server's console. 
If i don't exec one of subscription(ros2 topic echo or something) for the topic, i cannot see a echo lines in the console or the server's console.

```
New endpoint discovered: Endpoint{01.0f.2e.e2.5a.05.2e.5a.01.00.00.00|0.0.1.3;writer;Topic{ros_discovery_info;rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_;TopicQoS{Fuzzy{Level(40) 1};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.5a.05.2e.5a.01.00.00.00|0.0.3.3;writer;Topic{rt/rosout;rcl_interfaces::msg::dds_::Log_;TopicQoS{Fuzzy{Level(40) 1};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.5a.05.2e.5a.01.00.00.00|0.0.4.3;writer;Topic{rt/parameter_events;rcl_interfaces::msg::dds_::ParameterEvent_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.5a.05.2e.5a.01.00.00.00|0.0.2.4;reader;Topic{ros_discovery_info;rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_;TopicQoS{Fuzzy{Level(40) 1};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
DDS Router running.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.1.3;writer;Topic{ros_discovery_info;rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_;TopicQoS{Fuzzy{Level(40) 1};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.2.4;reader;Topic{ros_discovery_info;rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_;TopicQoS{Fuzzy{Level(40) 1};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.3.3;writer;Topic{rt/rosout;rcl_interfaces::msg::dds_::Log_;TopicQoS{Fuzzy{Level(40) 1};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.4.4;reader;Topic{rq/talker/get_parametersRequest;rcl_interfaces::srv::dds_::GetParameters_Request_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.5.3;writer;Topic{rr/talker/get_parametersReply;rcl_interfaces::srv::dds_::GetParameters_Response_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.6.4;reader;Topic{rq/talker/get_parameter_typesRequest;rcl_interfaces::srv::dds_::GetParameterTypes_Request_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.7.3;writer;Topic{rr/talker/get_parameter_typesReply;rcl_interfaces::srv::dds_::GetParameterTypes_Response_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.8.4;reader;Topic{rq/talker/set_parametersRequest;rcl_interfaces::srv::dds_::SetParameters_Request_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.9.3;writer;Topic{rr/talker/set_parametersReply;rcl_interfaces::srv::dds_::SetParameters_Response_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.a.4;reader;Topic{rq/talker/set_parameters_atomicallyRequest;rcl_interfaces::srv::dds_::SetParametersAtomically_Request_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.b.3;writer;Topic{rr/talker/set_parameters_atomicallyReply;rcl_interfaces::srv::dds_::SetParametersAtomically_Response_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.c.4;reader;Topic{rq/talker/describe_parametersRequest;rcl_interfaces::srv::dds_::DescribeParameters_Request_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{Level(0) 0});downsampling(Fuzzy{Level(0) 1})}(payload::rtps::v0)};SpecificEndpointQoS{Partitions{};OwnershipStrength{0}};Active;SimpleParticipant_domain0}.
New endpoint discovered: Endpoint{01.0f.2e.e2.66.27.0e.ec.01.00.00.00|0.0.d.3;writer;Topic{rr/talker/describe_parametersReply;rcl_interfaces::srv::dds_::DescribeParameters_Response_;TopicQoS{Fuzzy{Level(40) 0};Fuzzy{Level(40) 0};Fuzzy{Level(40) };depth(Fuzzy{Level(0) 5000});max_tx_rate(Fuzzy{Level(0) 0});max_rx_rate(Fuzzy{ [... cut ...]
    verbose: true                                                     # 10

  - name: WANServer                                               # 6
    kind: wan                                                     # 7
    listening-addresses:                                          # 8
      - ip: 123.123.123.123                                       # 9
        port: 11666                                               # 10
        transport: tcp                                            # 11
```

Woojin-Crive, 2023-12-05:
I found the Solution myself...

```
builtin-topics:
  - name: rt/chatter                                                  
    type: std_msgs::msg::dds_::String_
```

closed 2023-12-05 by Woojin-Crive

juanlofer-eprosima, 2023-12-12:
Hi @Woojin-Crive , as you managed to realize DDS-Router's entities for a topic are only created when discovering a reader in that same topic. However, this is the default behaviour and can be changed with [discovery trigger](https://eprosima-dds-router.readthedocs.io/en/latest/rst/user_manual/configuration.html#discovery-trigger) configuration option. If wishing to keep the default behaviour, builtin topics is indeed the way to go.