eProsima/DDS-Router #458: History queue not updated

yjaeger83, 2024-06-17:
Hello,

We are currently using ddsrouter with different Qos. We recently upgraded from version 1.0.0 to version 2.2.0 and we are facing an issue.
We did not change the consumer code (ddsparticipant using the "take" functions) but it seems that the values that have QoS set to transient local are kept inside the history (although the history policy is KEEP_LAST).
For example, if the history-depth is set to 5, our consuming component retrieves the 5 oldest elements (with Qos transient local) from the history and all new values seemed to be dropped and never make it to the calling code.

juanlofer-eprosima, 2024-06-17:
Hello @yjaeger83 , could you please provide a concrete example of messages sent by your publisher, messages expected to be received by your subscriber, and actual messages expected received by your subscriber? I'm afraid we were not able to fully understand the problem.
Also, could you please clarify what are the history and durability configurations for your publisher and subscriber?

yjaeger83, 2024-06-17:
Hello,
The history policy of our publisher and subscriber is KEEP_LAST with a history depth of 1.

yjaeger83, 2024-06-17:
We are only transferring text messages in our topics. When history-depth is set to 5 in the ddsRouters configuration, then for the topics that are set to be transient local, we will receive 5 updates of those topics, the subsequent updates of those topics are somehow dropped or ignored.
For example if topic1 is sent sequencially with following values "test1", "test2", "test3", "test4", "test5", "test6", the subscriber will only receive "test1", "test2", "test3", "test4", "test5". All subsequent values of topic1 are either not sent by the publisher or received by the subscriber.

juanlofer-eprosima, 2024-06-18:
Hi @yjaeger83 , thank you for the detailed information, we were able to reproduce the issue.
The fix will be introduced in the following [PR](https://github.com/eProsima/DDS-Pipe/pull/122) and included in the upcoming product release.

Cheers!

yjaeger83, 2024-06-18:
Thank you @juanlofer-eprosima for your quick reaction.

closed 2024-07-15 by juanlofer-eprosima