eProsima/DDS-Router #490: Difference in DDS Security Behavior Between Vulcanexus and Manual Build (DDS Router 3.1.0)

bubssi528, 2025-02-06:
I am using eProsima DDS Router (version 3.1.0) and have encountered an issue with DDS Security. 
When I source Vulcanexus using:

`source /opt/vulcanexus/humble/setup.bash`

DDS Security works as expected. However, when I manually build the latest version of DDS Router 3.1.0 from GitHub on Linux, DDS Security does not seem to apply.

I would like to understand what might be causing this difference. Are there specific build flags, dependencies, or configurations in Vulcanexus that enable DDS Security, which might be missing in my manual build?

vulcanexus test is success
[https://docs.vulcanexus.org/en/latest/rst/tutorials/cloud/secure_router/secure_router.html](url)

maul build ddsrouter official site 
[https://eprosima-dds-router.readthedocs.io/en/latest/rst/developer_manual/installation/sources/linux.html](url)

juanlofer-eprosima, 2025-02-06:
Hi @bubssi528 , did you compile Fast-DDS with `-DSECURITY=ON`? This is a requirement for security to function, and that's how it's done in Vulcanexus.

bubssi528, 2025-02-07:
I am currently using **FastDDS** for **ROS2**, so I did not explicitly build it with `-DSECURITY=ON`. Instead, I am applying DDS Security using environment variables as shown below:

```
export ROS_SECURITY_ENABLE=true
export ROS_SECURITY_STRATEGY=Enforce
export ROS_SECURITY_KEYSTORE=/path/to/keystore
```

juanlofer-eprosima, 2025-02-07:
@bubssi528 I don't quite understand what do you mean by "I am currently using FastDDS for ROS2". Are you building DDS-Router 3.1.0 against the Fast-DDS library already present in Vulcanexus? That should not be possible, as these versions are not compatible (see dependencies [here](https://github.com/eProsima/DDS-Router/releases/tag/v3.1.0)). I suggest you build DDS-Router together with all its dependent packages for that specific version (providing `-DSECURITY=ON` when building Fast-DDS).

Alternatively, if what you need is to use this particular DDS-Router, you may instead use Vulcanexus Jazzy which includes it already.

bubssi528, 2025-02-07:
sorry i found myisstake.
it's done

closed 2025-02-07 by bubssi528