eProsima/DDS-Router #502: DDS-Router Local Discovery Server not working

adamanasfi, 2025-04-29:
Working with ROS2. 
Following this documentation: https://eprosima-dds-router.readthedocs.io/en/latest/rst/examples/ros_discovery_server_example.html

First terminal: ROS_DOMAIN_ID=0 ros2 run demo_nodes_cpp talker (I see the Hello World with count being logged). I could echo this topic in another terminal.

Second terminal: ROS_DISCOVERY_SERVER=";127.0.0.1:11888" ros2 run demo_nodes_cpp listener

Third terminal: ./install/ddsrouter_tool/bin/ddsrouter --config-path /home/vrl/DDS-Router/install/ddsrouter_tool/share/resources/configurations/examples/ros_discovery_server.yaml
Output:
Starting DDS Router Tool execution.
DDS Router running.

yaml file is as follows (same as documentation):

##################################
# ALLOWED TOPICS
# Allowing ROS2 HelloWorld demo_nodes topic

allowlist:
  - name: rt/chatter                               # 1
    type: std_msgs::msg::dds_::String_             # 1

##################################
# PARTICIPANTS
participants:

##################################
# SIMPLE PARTICIPANT
# This participant will subscribe to topics in allowlist in domain 0 and listen every message published there

  - name: SimpleROS2                               # 2
    kind: local                                    # 3
    domain: 0                                      # 4

##################################
# ROS DISCOVERY SERVER
# This participant will subscribe to topics in allowlist using Discovery Server protocol as Server

  - name: ServerROS2                               # 5
    kind: local-discovery-server                   # 6
    listening-addresses:                           # 7
      - domain: localhost                          # 8
        port: 11888                                # 9
                                                   # 10

##################################
# CONFIGURATION DESCRIPTION

PROBLEM: Listener is not receiving any messages.

Please note that I am very new to networking and have very limited debugging skills in this area.

adamanasfi, 2025-04-29:
Realized I am running a server, when I was actually interested in a client.

juanlofer-eprosima, 2025-04-29:
Hi @adamanasfi , I could not reproduce the issue with DDS-Router v3.1.0. I tested the scenario using `eprosima/vulcanexus:jazzy-desktop` at Docker Hub. It might be a communication issue between different Docker containers if that's the way you are operating.

JTShuai, 2025-07-22:
Same here. Did you get any updates?

I tried the [change doman example](https://eprosima-dds-router.readthedocs.io/en/latest/rst/examples/change_domain_example.html) which works well. But this Local Discovery Server doesn't work.